exelban/stats #1319: Suggestion

AShortcuts, 2023-02-19:
It would be great if it could show the amount of RAM free and used when hovering over the RAM bar icon. I know that there is a menu bar widget for that, but this way it wouldn't take up unnecessary space on the menu bar.
Thank you.

[image]

exelban, 2023-02-19:
Hi. Thanks for the feedback.
I will take a look at this.

exelban, 2023-02-27:
Hi. So I take a look if it could be added.
And yes, it could be added. But. I can only duplicate the value from the widget. For example CPU usage. But I don't see any reason to duplicate the same value 2 times in one view. For example, the CPU widget will show 36% of usage, and I can show the same value in the tooltip. But there is no sense to do it. Because this value is already visible without waiting for the tooltip.

It's the main reason why the popup view is presented in the app. If you want to see more than in the widget just click and you will see more info about the usage.

closed 2023-02-27 by exelban