exelban/stats #1322: Energy usage

JonasPeitz, 2023-02-20:
Hey, is there a way to improve stats of energy usage?

exelban, 2023-02-20:
Hi. What do you mean?

JonasPeitz, 2023-02-21:
I use a MacBook Air m2 and wanted to use stats primarily for heat measurements. I noticed that it has a high-energy usage (what's kind of normal for monitoring services). Likewise, I read your notes now after my post and I disabled the sensor. Is there an option to only enable half of the sensors? Since my MacBook hasn't proper cooling, measuring temperatures is kind of necessary. The energy usage is kind of back to normal after I disabled the sensors.

exelban, 2023-02-21:
I'm trying to keep energy usage as low as it's possible. If you want to decrease the usage you can disable unused modules. There is no way to improve energy usage without removing features. If you want fine-tuned app just for you you can make a fork and disable parts that are not useful for you.

closed 2023-02-21 by exelban

JonasPeitz, 2023-02-21:
I am not good at swift programming, but maybe I will do it just for fun. I primarily need the temperature of my CPU and GPU. My hope is when I disable all other sensors, the power consumption will decrease. I wouldn't do it currently and I kind of have a feeling now for heat usage since I used your app already to test, compiling, rendering and co. Currently, I have an important test at school, so I wouldn't have time. 

Can you please send me a swift tutorial and the class for the sensors, I will try to implement simple buttons then.

exelban, 2023-02-21:
There are not so many differences if you are reading few sensors or all. It's not a simple task to read the value of a sensor. But I'm pretty sure I saw some library in python that allows fetching sensors on macOS. Maybe you can use them.