exelban/stats #1324: Wrong computer info

retmia, 2023-02-21:
I'm using a 16' MacBook Pro M2 Max but the software sees it as 14'
Regards

exelban, 2023-02-21:
Please share the output of `system_profiler SPHardwareDataType`.

retmia, 2023-02-21:
Here it is : 

      Model Name: MacBook Pro
      Model Identifier: Mac14,6
      Model Number: MNWA3FN/A
      Chip: Apple M2 Max
      Total Number of Cores: 12 (8 performance and 4 efficiency)
      Memory: 32 GB
      System Firmware Version: 8419.80.7
      OS Loader Version: 8419.80.7
      Serial Number (system): ----------
      Hardware UUID: ----------
      Provisioning UDID: ----------
      Activation Lock Status: Enabled
      
I hope you don't need the SN & UDIDs.
Regards

exelban, 2023-02-21:
Thanks. I just need a model identifier. Will be fixed.

retmia, 2023-02-21:
Thanks for your attention :)

exelban, 2023-02-21:
thanks for the feedback

closed 2023-02-26 by exelban