exelban/stats #1335: Fan speed shown as RPM but configured to show percentage

n-6, 2023-03-02:
I have configured a widget to show the speed of the fastest fan as a percentage (vs RPM)  However, the widget unexpectedly displays RPM.  The screenshot below shows a temperature of 99 degrees Celsius with a fan speed of 2500 below it, along with the Fan value configuration set to Percentage.

[image]

**Details:**
 - Device: Macbook Pro M1 Max
 - macOS: 13.2.1
 - Application version: 2.8.10

n-6, 2023-03-02:
If I display either the left fan speed or right fan speed, the percentage is shown.  Only when I ask for the fastest fan speed is the RPM shown instead of the percentage.

exelban, 2023-03-02:
ok, I will take a look.

closed 2023-03-12 by exelban

n-6, 2023-03-17:
Thanks!  Confirmed as fixed in v2.8.12.  Donation on the way.