exelban/stats #1336: Wrong 'Allow in the Background' name and icon

Atcold, 2023-03-02:
Going to  / System Settings… / General / Login Items I have the entry Serhiy Mytrovtsiy in the Allow in the Background list.
It should say something like Stats and show the correct icon instead.

[image]

For example, see the Microsoft AutoUpdate item.

[image]

**Details:**
 - Device: Macbook Pro 2023, Apple M2 Max
 - macOS: 13.2.1
 - Application version: 2.8.10

exelban, 2023-03-02:
hmm. I will need to check that.

exelban, 2023-03-07:
Hi. Ok, I take a look at that. And it cannot be changed. This item it's a fans helper xpc service. And it's added to the launch daemons with the developer account name. I could not find a way how that name could be changed.

closed 2023-03-07 by exelban