exelban/stats #1337: Dashboard displays incorrect laptop model (MacBook Pro 14", M2 Pro)

ritwiktakkar, 2023-03-03:
**Describe the bug**
The dashboard incorrectly shows my MacBook Pro 14" (M2 Pro) as a MacBook Pro 16" (M2 Pro). [image] The window on the left is the Stats Dashboard and the window on the right is →About This Mac.

**Details:**
 - Device: MacBook Pro (14-inch, 2023)
 - macOS: 13.2.1
 - Application version: 2.8.10

exelban, 2023-03-03:
Please share the output of system_profiler SPHardwareDataType.

ritwiktakkar, 2023-03-03:
Please see below (I removed the SN, Provisioning UDID, and Hardware UUID; please let me know if UUID is needed):

Hardware:

    Hardware Overview:

      Model Name: MacBook Pro
      Model Identifier: Mac14,9
      Model Number: MPHH3LL/A
      Chip: Apple M2 Pro
      Total Number of Cores: 10 (6 performance and 4 efficiency)
      Memory: 16 GB
      System Firmware Version: 8419.80.7
      OS Loader Version: 8419.80.7
      Activation Lock Status: Disabled

exelban, 2023-03-03:
Thanks, will be fixed.

closed 2023-03-05 by exelban