exelban/stats #1338: [Feature Request] Support spacing characters

dmlemos, 2023-03-03:
I saw that other requests regarding spacing were closed as not planned, and I wanted to suggest the ability to define characters for each space. It would be easier to identify the properties one is looking for.

Example of this in ASCII:
```
[CPU RAM] SSD (BATTERY)
```
The characters `[`, `]`, `(`, `)` would be used as the spacing.

exelban, 2023-03-03:
`other requests regarding spacing were closed as not planned ` - if you have to look at a few of them you will understand that it's impossible to change. Because it's a system add spacing, and it cannot be changed.
According to the symbols, I totally do not understand what you propose. Please explain more.

dmlemos, 2023-03-03:
Sorry I should have explained a bit more. The idea was to group different stats together and at the same time close the gap/spacing, but on a second thought it probably wouldn't change the spacing.

exelban, 2023-03-04:
There is an option to group the widgets of one module. Or group the all modules in one view. It's not enough?

closed 2023-03-04 by exelban

dmlemos, 2023-03-05:
I got another idea which I believe makes more sense for what I'm trying to achieve.
When modules are combined, the popup view is replaced by a view without much information. I'd love to save space but also have the details available when needed. With that said:
1. Is it feasible to implement a detailed view of each module to appear in the combined view popup? perhaps as an expanding list - this would be the ideal. Advantage of saving a lot of space with the information needed
1. Is it feasible to combine only certain modules, and leave the rest ungrouped? less ideal. Advantage of saving only some space but more messy

I realise this is a different request, just putting my thoughts out there. Thanks for your patience :)