exelban/stats #1344: Stats's Custom Update Routine Breaks App Installation When Installed/Run by Non-Admin User

gurple, 2023-03-06:
When the Stats app is installed by a non-admin account to the /Applications directory its ownership and permission modes remain with that non-admin. However, the non-admin account cannot modify the top level of the /Applications directory.

When the custom Stats update process is triggered while running under the non-admin user's session the process fails half way because the non-admin does not have the rights to modify the contents of /Applications. This leaves in /Applications a mangled and unusable /Applications/Stats.app/ structure.

An enhanced logic is needed to accommodate uses under non-admin accounts. Consider using the Sparkle framework which has already addressed this issue.

exelban, 2023-03-07:
Sorry but I don't see a reason why someone needs to run this application under non-admin account. If you want to have updates to need to have permission to your application folder. Otherwise, you can do a manual update. If you have a proposal how non user could receive an update I will be glad to implement that. But for sure I will not spend my time on that case.

gurple, 2023-03-08:
Software should not nuke itself due to its assumptions. It should first validate before taking destructive steps. A write test should occur. If the write test fails on app's base directory then it should not take the next steps of trying to write. This is a design bug and has nothing to do with "dumb user doing it wrong".

gurple, 2023-03-12:
I can't really test it but here's an 'updater.sh' version that might be a little more robust in checking rights and respecting return codes. Consider it as food-for-thought.
Cheers
--------------------------------------------------------------------------------

#!/bin/bash
EXIT_STATUS=1 #Set non-zero to use as default to encourage explicitly setting

STATS_DMG_PATH="{$HOME}/Downloads/Stats.dmg"
STATS_DMG_MOUNT_PATH="/tmp/Stats"
GLOBAL_APPLICATIONS_PATH="/Applications"

STEP=""

while [[ "$#" -gt 0 ]]; do case $1 in
  -s|--step) STEP="$2"; shift;;
  -d|--dmg) STATS_DMG_PATH="$2"; shift;;
  -a|--app) GLOBAL_APPLICATIONS_PATH="$2"; shift;;
  -m|--mount) STATS_DMG_MOUNT_PATH="$2"; shift;;
  *) echo "Unknown parameter passed: $1" ; exit "$EXIT_STATUS";;
esac; shift; done

if [ "$(id -nG | grep -q 'admin' ; echo "$?")" -ne 0 ] ; then
	echo "You must run this under a macOS admin account."
	exit "$EXIT_STATUS"
fi

if [[ "$STEP" -eq "2" ]]; then
	if [ -w "$GLOBAL_APPLICATIONS_PATH" ] ; then
		rm -rf "$GLOBAL_APPLICATIONS_PATH"/Stats.app 
		if [ "$(echo "$?")" -eq 0 ] ; then
			cp -rf "$STATS_DMG_MOUNT_PATH"/Stats.app "$GLOBAL_APPLICATIONS_PATH"/Stats.app
		else
			echo "Failed to remove $GLOBAL_APPLICATIONS_PATH/Stats.app"
			exit "$EXIT_STATUS"
		fi
		if [ "$(echo "$?")" -eq 0 ] ; then
			 "$GLOBAL_APPLICATIONS_PATH"/Stats.app/Contents/MacOS/Stats --dmg "$STATS_DMG_PATH" && echo "New version started"
		fi
	else
		echo "Cannot write to the $GLOBAL_APPLICATIONS_PATH. Halting."
		exit "$EXIT_STATUS"
	fi
    
elif [[ "$STEP" -eq "3" ]]; then
    /usr/bin/hdiutil detach "$STATS_DMG_MOUNT_PATH" 
	if [ "$(echo "$?")" -eq 0 ] ; then
		/bin/rm -rf "$STATS_DMG_MOUNT_PATH"
	else
		echo "Failed to unmount $STATS_DMG_MOUNT_PATH"
		exit "$EXIT_STATUS"
	fi
	if [ "$(echo "$?")" -eq 0 ] ; then
		/bin/rm -rf "$STATS_DMG_PATH" && echo "Done"
	else
		echo "Failed to remove temp mount directory $STATS_DMG_MOUNT_PATH"
		exit "$EXIT_STATUS"
	fi

else
    rm -rf "$GLOBAL_APPLICATIONS_PATH"/Stats.app
	if [ "$(echo "$?")" -eq 0 ] ; then
		cp -rf "$STATS_DMG_MOUNT_PATH"/Stats.app "$GLOBAL_APPLICATIONS_PATH"/Stats.app
	else
		echo "Failed to remove $GLOBAL_APPLICATIONS_PATH/Stats.app
		exit "$EXIT_STATUS"
	fi
	if [ "$(echo "$?")" -eq 0 ] ; then
		 "$GLOBAL_APPLICATIONS_PATH"/Stats.app/Contents/MacOS/Stats --dmg-path "$STATS_DMG_PATH" --mount-path "$STATS_DMG_MOUNT_PATH" && echo "New version started."
	else
		echo "Failed to copy $STATS_DMG_MOUNT_PATH/Stats.app to $GLOBAL_APPLICATIONS_PATH/Stats.app"
		echo "$EXIT_STATUS"
	fi
fi
EXIT_STATUS=0
exit "$EXIT_STATUS"

gurple, 2023-03-13:
I would strongly encourage transitioning to and integrating the Sparkle update framework: https://sparkle-project.org/ here.

It would help avoid a lot of potential pain points that can arise using these custom shell scripts and hard coded app names and paths and managing the downloads of new Stats.dmg files, creating temp scratch directories, etc. There are a lot of dragons out in these waters.

exelban, 2023-03-13:
@gurple Sparkle will be never used.

closed 2023-03-19 by exelban