exelban/stats #1347: On Mac Mini M2 Pro (2023) Power Consumption displays „denied symbol“

dirk68-fu, 2023-03-07:
On an Mac Mini M2 Pro (2023) the Power Consumption Widget show the 🚫 symbol instead of a value.
The Sensor Dropdown menu shows correct values, but the widget does not.

exelban, 2023-03-07:
Because you have no enabled sensors.

closed 2023-03-07 by exelban

dirk68-fu, 2023-03-07:
Why can't the values from the dropdown menu be displayed in the menu bar? They seem to be correct.

exelban, 2023-03-07:
because you need to enable them

dirk68-fu, 2023-03-07:
I did enable one of them and it still shows 🚫.
I don't understand your answer: Is this not supported for my model or did I not enable it correctly.
Im pretty sure I switched at least one switch in and it still showed 🚫.

exelban, 2023-03-07:
please share the screenshot of the sensors module settings

dirk68-fu, 2023-03-09:
[image]

dirk68-fu, 2023-03-09:
Thank you for your patience.
It works now, but I don't know why it didn't before. I had to remove the sensor widget and reenable it to make it work.