exelban/stats #1348: Stats is preventing sleep on Mojave

minimum91, 2023-03-08:
Hi there, thank you for the extremely useful and customisable utility.
Is there anything I can do on my end to stop it preventing sleep?

I am running 10.14.6 on 5,1 Mac Pro.

[image]

BitesPotatoBacks, 2023-03-08:
@minimum91 I don't know what's going on there but we could try and narrow down what assertions Stats is making to prevent sleep. What is your output from running `pmset -g assertions` in Terminal?

minimum91, 2023-03-08:
@BitesPotatoBacks 

Here's the output from the command:

2023-03-09 00:23:34 +0200 
Assertion status system-wide:
   BackgroundTask                 1
   ApplePushServiceTask           0
   UserIsActive                   1
   PreventUserIdleDisplaySleep    0
   PreventSystemSleep             0
   ExternalMedia                  0
   PreventUserIdleSystemSleep     1
   NetworkClientActive            0
Listed by owning process:
   pid 84(apsd): [0x0000419a000ba625] 00:00:17 ApplePushServiceTask named: "com.apple.apsd-waitingformessages-push.apple.com" 
   pid 107(hidd): [0x0000372f00098940] 00:00:00 UserIsActive named: "com.apple.iohideventsystem.queue.tickle.4294976813.3" 
	Timeout will fire in 900 secs Action=TimeoutActionRelease
   pid 285(UserEventAgent): [0x0000415d000ba54c] 00:01:18 BackgroundTask named: "eu.exelban.Stats.Network.IP" 
	Created for PID: 14032. 
   pid 333(sharingd): [0x000041150001a479] 00:02:30 PreventUserIdleSystemSleep named: "Handoff" 
Kernel Assertions: 0x10c=USB,BT-HID,MAGICWAKE
   id=500  level=255 0x4=USB mod=01/01/1970, 03:00 description=com.apple.usb.externaldevice.fd300000 owner=Duet USB
   id=502  level=255 0x4=USB mod=01/01/1970, 03:00 description=com.apple.usb.externaldevice.1d100000 owner=flirc
   id=503  level=255 0x4=USB mod=01/01/1970, 03:00 description=com.apple.usb.externaldevice.1a200000 owner=IOUSBHostDevice
   id=505  level=255 0x4=USB mod=01/01/1970, 03:00 description=com.apple.usb.externaldevice.1a230000 owner=USB Receiver
   id=506  level=255 0x100=MAGICWAKE mod=08/03/2023, 23:50 description=en0 owner=en0
   id=507  level=255 0x100=MAGICWAKE mod=08/03/2023, 23:46 description=en2 owner=en2
   id=534  level=255 0x8=BT-HID mod=01/01/1970, 03:00 description=com.apple.driver.IOBluetoothHIDDriver owner=AppleHSBluetoothDevice
   id=535  level=255 0x8=BT-HID mod=01/01/1970, 03:00 description=com.apple.driver.IOBluetoothHIDDriver owner=AppleHSBluetoothDevice
Idle sleep preventers: IODisplayWrangler

BitesPotatoBacks, 2023-03-08:
Well the `BackgroundTask` assertion is clearly `1` (true), which corresponds with this line:
```
pid 285(UserEventAgent): [0x0000415d000ba54c] 00:01:18 BackgroundTask named: "eu.exelban.Stats.Network.IP"
Created for PID: 14032.
```
From what I recall, Stats communicates with exelban's server in order to retrieve user IP, and this could be what's blocking sleep....

@minimum91 do you have the network module enabled in Stats? Perhaps disabling that module could unblock sleep

exelban, 2023-03-09:
Yeap. there you have a process that wakes up your mac: 
`pid 333(sharingd): [0x000041150001a479] 00:02:30 PreventUserIdleSystemSleep named: "Handoff"`

There is nothing in the logs you shared saying that Stats prevents mac from sleeping.

closed 2023-03-09 by exelban

minimum91, 2023-03-09:
> Well the `BackgroundTask` assertion is clearly `1` (true), which corresponds with this line:
>
> ```
> pid 285(UserEventAgent): [0x0000415d000ba54c] 00:01:18 BackgroundTask named: [...]

Seems to be off, if this is what you are referring to. I just use one RAM block, that's all.

[image]
[image]

> Yeap. there you have a process that wakes up your mac: `pid 333(sharingd): [0x000041150001a479] 00:02:30 PreventUserIdleSystemSleep named: "Handoff"`
> 
> There is nothing in the logs you shared saying that Stats prevents mac from sleeping.

I know it's a bit counter-intuitive, but Handoff has always been there on my machine, yet it never prevented the Mac from sleeping. As soon as I quit Stats, the machine goes to sleep as normal.

exelban, 2023-03-09:
I have running Stats on a few macs and never cause the problem with sleep. Also, I'm pretty sure that the network request could not prevent mac from sleeping.

If you have disabled the network module it means only one: the request was never done. Just the task was registered. So Stats was not even sent a request.

exelban, 2023-03-09:
@BitesPotatoBacks so you want to say that `NSBackgroundActivityScheduler ` prevents sleep?

BitesPotatoBacks, 2023-03-09:
@exelban Yes based on what I could test that seems to be the case... Though, this sleep problem isn't an issue on my Mac Mini (which is running Ventura) so perhaps this sleep prevention is just a problem with the way Mojave might handle background tasks?

exelban, 2023-03-09:
It's very strange. If you will debug why it's happening and how it could be fixed ping me.
I cannot reproduce that bug on a macbook or mac studio. Also before have a hackintosh, and even there I have no problems with sleep.

BitesPotatoBacks, 2023-03-09:
@exelban perhaps it could be fixed by invalidating the `ipUpdater` when the Network module is disabled, and then rescheduling with `setIPUpdater()` on re-enable? You could make the `enable()` and `disable()` Module methods open, that way you could override them in the Network subclass and do the invalidation/rescheduling from within those overrides. 

I believe that could solve this problem, but like I said, I cannot recreate the sleep prevention, I only observed the persistence of the background activity. So If those changes were made, @minimum91 would have to test the new build in order to validate my guessing. 🤔

exelban, 2023-03-10:
Stats already has it