exelban/stats #1350: Some issues of Power stats Average system total

Aragamiz, 2023-03-09:
In my Power stats Average system total
It's lock 14.38w
I'm useing m1 pro+12.6.3 Mac
It's bug?

Aragamiz, 2023-03-10:
> what value has the `System Total` sensor?

The power consumption of system total is displayed in time, which changes with my use. But Average system total is not. I used it for a long time yesterday and displayed 14.38w, and turned it on this morning and displayed 13.65w. Is Average system total the system value read, or is it obtained through a set of stats algorithm?

exelban, 2023-03-10:
It's calculated inside the stats: Average System Total = Total System Consumption * 3600 / sinceFirstRead

Aragamiz, 2023-03-10:
> It's calculated inside the stats: Average System Total = Total System Consumption * 3600 / sinceFirstRead

Can add a refresh button and time since calculated? This let me get some confuse...
thks a lot~

exelban, 2023-03-10:
nope, there is no option to reset that.

closed 2023-03-10 by exelban