exelban/stats #1351: Suggestion: increase text contrast and adapt to Dark Mode

andrecasal, 2023-03-10:
**Describe the bug**
It would be great if the text colors had higher contrast and adapted to Dark Mode

The RAM % is hard to see:
[image]

And here it's practically invisible:
[image]

**Details:**
 - Device: MacBook Pro 2019 16"
 - macOS: Ventura 13.1
 - Application version: 2.8.11

exelban, 2023-03-12:
It's not related to the contrast. If I make the text a little bit darker someone who has a little bit of dark wallpaper will come and ask to do the same.

closed 2023-03-12 by exelban

andrecasal, 2023-03-14:
Thank you for your response. The menu bar icons in macOS are designed to be adaptive and change color based on the background color, ensuring that they remain visible and easily identifiable regardless of the user's wallpaper.

Check it out:
[image]
[image]

I think adapting the text color would be a valuable addition to Stats.app.