exelban/stats #1352: App crashing in the background

Smultie, 2023-03-11:
**Describe the bug**
I often return to my MacBook to find stats.app no longer running. No error message onscreen. Need some help bugtracking (any logs somewhere, etc?)

**Details:**
 - Device: MacBook Air M2 base
 - macOS: 13.2.1
 - Application version: 2.8.11

Smultie, 2023-03-11:
Not sure if you want me to just post it here. 
If not, let me know how to get the info to you in another way.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [2130]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.8.11 (464)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-03-11 10:11:00.1542 +0100
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        41B3BEBF-AEDF-DF95-4829-4A43C2798EF2

Sleep/Wake UUID:       47B54F36-2D10-4234-A076-363EC2AE0018

Time Awake Since Boot: 11000 seconds
Time Since Wake:       10579 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000759b3b803420
Exception Codes:       0x0000000000000001, 0x0000759b3b803420

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2130]

VM Region Info: 0x759b3b803420 is not in any region.  Bytes after previous region: 23755925500961  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)
--->  
      UNUSED SPACE AT END

Kernel Triage:
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	       0x19c5f7b80 objc_retain + 16
1   libswiftCore.dylib            	       0x1aab104a8 swift_bridgeObjectRetain + 44
2   Kit                           	       0x103332258 0x1032a0000 + 598616
3   Kit                           	       0x103331be8 0x1032a0000 + 596968
4   Kit                           	       0x103331f6c 0x1032a0000 + 597868
5   AppKit                        	       0x19fde39c4 _NSViewDrawRect + 124
6   AppKit                        	       0x1a04965c8 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
7   AppKit                        	       0x19fde3238 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
8   QuartzCore                    	       0x1a40f9f40 CABackingStoreUpdate_ + 224
9   QuartzCore                    	       0x1a4155f6c invocation function for block in CA::Layer::display_() + 64
10  QuartzCore                    	       0x1a40f9118 -[CALayer _display] + 1644
11  AppKit                        	       0x19fde2f40 -[_NSBackingLayer display] + 352
12  AppKit                        	       0x19fd74968 -[_NSViewBackingLayer display] + 532
13  AppKit                        	       0x19fd74714 -[_NSBackingLayer displayIfNeeded] + 52
14  QuartzCore                    	       0x1a41517d0 -[CALayer renderInContext:] + 140
15  QuartzCore                    	       0x1a415347c -[CALayer _renderSublayersInContext:] + 340
16  QuartzCore                    	       0x1a4151bf8 -[CALayer renderInContext:] + 1204
17  QuartzCore                    	       0x1a415347c -[CALayer _renderSublayersInContext:] + 340
18  QuartzCore                    	       0x1a4151bf8 -[CALayer renderInContext:] + 1204
19  QuartzCore                    	       0x1a415347c -[CALayer _renderSublayersInContext:] + 340
[... 343 lines cut ...]
    "queue" : "com.apple.main-thread"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "60186475825c62000ccf5450",
      "factorPackIds" : {

      },
      "deploymentId" : 240000026
    },
    {
      "rolloutId" : "60f8ddccefea4203d95cbeef",
      "factorPackIds" : {
        "NETWORK_SERVICE_PROXY_CONFIG_UPDATE" : "6401414cc679dd6bc5828183"
      },
      "deploymentId" : 240000025
    }
  ],
  "experiments" : [

  ]
}
}

exelban, 2023-03-11:
Much better to share the file)

exelban, 2023-03-11:
What modules and widgets do you have enabled?

Smultie, 2023-03-11:
[Archive.zip](https://github.com/exelban/stats/files/10948545/Archive.zip)

There should be 3 logs in there.

Thanks

Smultie, 2023-03-11:
> What modules and widgets do you have enabled?

Can you find that from the logs, or want me to write them all down?

exelban, 2023-03-11:
Nope, not needed. I see where is a problem. Could you disable the network module and use the app for some time without that module? Looks like widget in this module causing the crash.

Smultie, 2023-03-11:
> Nope, not needed. I see where is a problem. Could you disable the network module and use the app for some time without that module? Looks like widget in this module causing the crash.

Just disabled it. If you receive no feedback, consider that module the cause :)

Smultie, 2023-03-11:
Network only. I don't have the Disk module enabled.

exelban, 2023-03-11:
ok, so it's a network module. So please give to know when you will be sure that Stats is not crashing anymore.

Smultie, 2023-03-11:
Will do.

!remindme 3 days :)

exelban, 2023-03-13:
hi. Thanks.

exelban, 2023-03-20:
Hi. Please try this build with an enabled network module.
[Stats.dmg.zip](https://github.com/exelban/stats/files/11021483/Stats.dmg.zip)

Smultie, 2023-03-20:
[image]
[image]

Testing atm!

exelban, 2023-03-21:
Hm, but you don't have enabled network chart

Smultie, 2023-03-21:
Nope, never had... Still crashed tho

exelban, 2023-03-21:
hmm, it's interesting. Please share the crash report from this version.

closed 2023-03-30 by exelban