exelban/stats #1353: Increase useability of "Combined modules" feature

Smultie, 2023-03-11:
**Describe the bug**
Not really a bug, more a feature request. 
I really like the information the app gives when clicking on the RAM (Top Processes, Memory pressure, etc) or CPU portion (different cores, average load, Top processes).
I also really like the aesthetics of the Combined Modules feature.

However, whenever I use Combined Modules, I can no longer use the extra information mentioned above (Top Processes, Memory pressure, etc).

Would there be any way to combine it into the best of both worlds? So, the nice way it looks in the menu bar, but also the possibility to click on CPU or RAM to show that specific information?

**Details:**
 - Device: MBA M2 2022
 - macOS: 13.2.1
 - Application version: 2.8.11

exelban, 2023-03-12:
There will be no changes in the nearest future. I have no time to redesign the popup view. Maybe in the next year, it will be developed, but for sure not in the near year.

closed 2023-03-12 by exelban

iKrushYou, 2023-03-16:
Coming from iStats Menu. I love the aesthetic of Stats share the same sentiment as @Smultie 
I would prefer to use the combined view but it would be nice to have drill-in buttons in the combined popup to just show the popup that would have been displayed in the individual mode views 

Also, let me know if I should file this separately, but I would love the option to choose which modules show up in the combined view. For example, I'd like to have battery separate but have the others still combined