exelban/stats #1360: Some widgets don't show up on mac startup

nicolopadovan, 2023-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

**Details:**
 - Device: Macbook Pro 2021, M1 Max
 - macOS: Ventura 13.2.1
 - Application version: 2.8.12

Most widgets, on mac startup, are not visibile - even though the widgets are active in the settings.
Toggling them off and back on solves the issue, but this must be done every time I turn the computer on.
The problem may be related to the fact that I am using an ultra wide monitor, and maybe the app is calculating the available horizontal space based on the macos built in retina screen (which is 16", but the space is limited due to the notch).

exelban, 2023-03-18:
What modules are not enabled on startup?
And what exactly do you need to toggle in the settings, the module toggle, or the widget itself?

nicolopadovan, 2023-03-18:
> What modules are not enabled on startup? And what exactly do you need to toggle in the settings, the module toggle, or the widget itself?

[image]

I need to toggle this switch off and on again, for each widget (RAM, CPU, GPU...)
While I don't think it's related to which toggle, but how far it is from the right side of the screen, the modules I have been experiencing the problem with are:
- CPU
- GPU
- RAM
- DISK

This doesn't appear to be happening to:
- Sensors
- Network
- Battery

Network and battery are the farthest right modules I have, but sensors is actually the farthest left (which suggests that either the sensors module is not being affected or that the position is not related to the problem.
[image]
Screenshot of my menu bar for completeness

exelban, 2023-03-19:
Please disable all modules except the CPU. Leave one active widget for the CPU module. Restart the application and check if the widget will be enabled.

nicolopadovan, 2023-03-19:
> Please disable all modules except the CPU. Leave one active widget for the CPU module. Restart the application and check if the widget will be enabled.

I think I mismatched between widget and module in my previous messages. 
The problem is not with "control center" widgets (or better: it may be, but I don't use the control center at all), but only in the menu bar icons.
Would the solution you provided be effective nonetheless?

exelban, 2023-03-19:
widget - it's an element in the menu bar.

nicolopadovan, 2023-03-20:
> Please disable all modules except the CPU. Leave one active widget for the CPU module. Restart the application and check if the widget will be enabled.

I disabled all modules except the CPU - restarting the app, the widget is in fact enabled.

kaspy003, 2023-03-22:
Yes, i have same problem.. every time when i start my mac studio (m1 max) i need to open app and toggle CPU checkbox to enable it ;(

exelban, 2023-03-22:
@kaspy003 please see my previous comment)

nicolopadovan, 2023-03-23:
> > Please disable all modules except the CPU. Leave one active widget for the CPU module. Restart the application and check if the widget will be enabled.
> 
> I disabled all modules except the CPU - restarting the app, the widget is in fact enabled.

Tried this fix, but it didn't work, and it appears that the problem is actually related to how far right each widget is: I tried changing the order, and the first on the right (in this case SSD) does show up, the others need to be toggled again.
Again, this doesn't effect sensors.

[image]

exelban, 2023-03-23:
please remove all elements from the menu bar and try again. Looks like you are trying to show more elements in the menu bar than it can fit.

nicolopadovan, 2023-03-23:
> please remove all elements from the menu bar and try again. Looks like you are trying to show more elements in the menu bar than it can fit.

It's true, to some extent: I am trying to show more elements than the menu bar on the Retina Display can fit; however, my other monitor does have the space to show all of the items.
Probably, the widgets are being added as soon as the mac is logged in, and while it's still refreshing the available monitors.
A possible fix would be to reload the widgets after this process has finished.

exelban, 2023-03-23:
Another. When you have enough space on your main screen do the widgets shows normally on start?

nicolopadovan, 2023-03-24:
> When you have enough space on your main screen do the widgets shows normally on start?

It seems so, yes.