exelban/stats #1363: Remove graph for Total System Consumption

LostInCompilation, 2023-03-19:
**Describe the bug**
If possible I think it would be better to remove the graph for the "Total System Consumption" value in the pop up. Since this value will always rise, the graph will always be at 100%. Removing it removes unnecessary information.

[image]

**Details:**
 - Device: MacBook Pro 2021 16"
 - macOS: 13.2.1
 - Application version: 2.8.12

exelban, 2023-03-19:
If you want to remove this value from the popup view there is an option to do that in the new version.

closed 2023-03-19 by exelban

LostInCompilation, 2023-03-19:
No I didn't mean the value itself. Only the graph when clicking on the value. The value itself is fine.