exelban/stats #1365: Fan sync controls bypass the lower boundary

StevenLi-phoenix, 2023-03-21:
When using Sensor module and controling fan speed, and enable the sync controling fan speed, the fan speed could be set below the minimum fan speed and cause no effects.

**Recreate:**
- enable fan speed control
- enable synchronous control 
- enable turbo mode
- wait for fan speed reach near 100%
- enable manunal mode
- move the slidbar of the first fan to minimum
- wait for sec and the first fan speed lowered to minimum but the second fan speed remain 100%

**Guessing:**
- In control fan speed moudle, it only check for if set value exceed maximum value but did not check the lower boundary
https://github.com/exelban/stats/blob/b26c555110b43e8213ad4c52ddff91087a4911f6/Modules/Sensors/popup.swift#L690

**Details:**
 - Device: MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports)
 - macOS: Monterey 12.6.3 (21G419)
 - Application version: 2.8.13
 
note: I got two fans with different speed. ([1250,6336],[1350,6864])

exelban, 2023-03-21:
Hi. Thanks for the feedback. Yeap, it's a bug. Will be fixed.

StevenLi-phoenix, 2023-03-21:
changing
```swift
 let value = sender.doubleValue < self.fan.maxSpeed ? sender.doubleValue : self.fan.maxSpeed 
```
to
```
 let value = sender.doubleValue < self.fan.maxSpeed ? (sender.doubleValue > self.fan.minSpeed ? sender.doubleValue :  self.fan.minSpeed ) : self.fan.maxSpeed

closed 2023-03-26 by exelban