exelban/stats #1370: [Feature Request]

bafulton, 2023-03-24:
Would it be possible to add the top `n` processes to the disk usage widget? It'd make it really convenient to know which process is causing my read/write rate to spike.

exelban, 2023-03-24:
unfortunately no, there is no an API for that.

bafulton, 2023-03-24:
Good to know. I wonder how iStat Menus does it. Maybe some sort of hacky workaround. 😬

[image]

BitesPotatoBacks, 2023-03-25:
@bafulton @exelban Actually, I think you should be able to get the read and writes of a process using `proc_pid_usage()`
```c
pid_t pid = 571; // some pid
rusage_info_current rusage;
    
if (proc_pid_rusage(pid, RUSAGE_INFO_CURRENT, (void **)&rusage) == 0) {
    printf("read: %llu write: %llu \n",
                rusage.ri_diskio_bytesread, // process reads
                rusage.ri_diskio_byteswritten); // process writes
}
```
Then just convert to bytes or whatever

exelban, 2023-03-25:
@BitesPotatoBacks hmm, it's could be a solution. Will take a look after vacation.

closed 2023-04-09 by exelban