exelban/stats #1372: Stats crashed upon boot

valynor, 2023-03-25:
Just booted up the Mac Studio, Stats icons appeared in the menu bar and ~2 seconds later it crashed. 
This is the first time this happened.

Crash log: 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [719]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.8.13 (466)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-03-25 14:32:37.5264 +0100
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        4016461C-24F3-2405-3768-F72B593C517E

Time Awake Since Boot: 30 seconds

System Integrity Protection: enabled

Crashed Thread:        6  Dispatch queue: eu.exelban.Stats

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000bd32f3501760 -> 0x00003d32f3501760 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000bd32f3501760

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [719]

VM Region Info: 0x3d32f3501760 is not in any region.  Bytes after previous region: 66808003434337  Bytes before following region: 38264076495008
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x196eca050 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x196edc11c mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x196ed2c24 mach_msg_overwrite + 452
3   libsystem_kernel.dylib        	       0x196eca3cc mach_msg + 24
4   CoreFoundation                	       0x196fe9bb0 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x196fe849c __CFRunLoopRun + 1232
6   CoreFoundation                	       0x196fe7878 CFRunLoopRunSpecific + 612
7   HIToolbox                     	       0x1a06c7fa0 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1a06c7de4 ReceiveNextEventCommon + 672
9   HIToolbox                     	       0x1a06c7b2c _BlockUntilNextEventMatchingListInModeWithFilter + 72
10  AppKit                        	       0x19a26d84c _DPSNextEvent + 632
11  AppKit                        	       0x19a26c9dc -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 728
12  AppKit                        	       0x19a260e0c -[NSApplication run] + 464
13  Stats                         	       0x102d829c8 0x102d80000 + 10696
14  dyld                          	       0x196bdfe50 start + 2544

Thread 1::  Dispatch queue: com.apple.root.background-qos
0   libsystem_kernel.dylib        	       0x196ecac0c read + 8
1   Foundation                    	       0x197ec1ea8 _NSReadFromFileDescriptorWithProgress + 192
2   Foundation                    	       0x197ee18dc -[NSConcreteFileHandle readDataOfLength:] + 368
3   Net                           	       0x103074984 0x103064000 + 67972
4   Kit                           	       0x103228788 0x1031c0000 + 427912
5   Kit                           	       0x1031caa18 0x1031c0000 + 43544
6   libdispatch.dylib             	       0x196d809dc _dispatch_call_block_and_release + 32
7   libdispatch.dylib             	       0x196d82504 _dispatch_client_callout + 20
8   libdispatch.dylib             	       0x196d940b4 _dispatch_root_queue_drain + 684
9   libdispatch.dylib             	       0x196d947c0 _dispatch_worker_thread2 + 164
10  libsystem_pthread.dylib       	       0x196f060c4 _pthread_wqthread + 228
[... 416 lines cut ...]
    "queue" : "eu.exelban.Stats"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000255
    },
    {
      "rolloutId" : "5f72dc58705eff005a46b3a9",
      "factorPackIds" : {

      },
      "deploymentId" : 240000015
    }
  ],
  "experiments" : [

  ]
}
}

exelban, 2023-03-25:
[report.txt](https://github.com/exelban/stats/files/11069039/report.txt)

exelban, 2023-03-25:
Looks like some system error. Error when creating `NSStackView`.

closed 2023-03-25 by exelban