exelban/stats #1374: [Feature Request] Control disk popup colors

bafulton, 2023-03-25:
Would it be possible to customize the upload/download colors in the disk usage widget's popup, similar to how the colors can be customized in the network usage widget (picture below)?

[image]

exelban, 2023-03-25:
yes, you can do it in the widget settings.

bafulton, 2023-03-30:
You can change the widget settings, but not the popup settings.

[image]

This means that if you customize the disk read/write colors in the menu bar, the colors in the popup won't match. In the screenshot below, I've set the menu bar's colors to green and blue, but in the popup it's still red and blue.

[image]

exelban, 2023-03-30:
hmm, I was sure that I have add that. Will be added.

bafulton, 2023-03-30:
Thank you! This app is so great--it's definitely in my top 5 must-have MacOS apps! 🥳

closed 2023-04-09 by exelban