exelban/stats #1375: battery status in menu bar shows percentage only instead of two lines

maia, 2023-03-26:
**Describe the bug**
When configuring the battery module to display two lines in the menu bar (percentage and time left to fully charge/discharge), it only displays percentage as a single line. See attached screenshot.

**Details:**
 - Device: MacBook Air M1
 - macOS: 13.2.1
 - Application version: 2.8.13
[image]

exelban, 2023-03-30:
you need to change the mode in the widget settings.

maia, 2023-03-30:
Thanks, I still need to get used to the three tabs (module, widget, popup) that each contain parts of the settings and change their content based on which widgets are "activated".