exelban/stats #1376: App is always visible in dock and app switcher

maia, 2023-03-27:
**Describe the bug**
The app is always visible in dock and app switcher (command-tab), even with the config window closed. I'd prefer, if it was "invisible" to dock and app switcher, like all other menu-bar-only-apps. One can also command-tab to it and then it has an empty menu, see attached screenshot.

**Details:**
Device: MacBook Air M1
macOS: 13.2.1
Application version: 2.8.13
[image]

exelban, 2023-03-30:
if you have no opened windows the application will not be available in the dock (only if you have enabled this option).

closed 2023-03-30 by exelban

maia, 2023-03-30:
Thanks, I didn't realise that option existed.