exelban/stats #1378: Stats is making outbound network connections even with auto-updates and connectivity checks disabled

mariobrostech, 2023-03-28:
**Describe the bug**
Stats, despite the options to check for updates and connectivity checks both being disabled, is making constant outbound connections to your website at [serhiy.io](https://serhiy.io). This isn't stated anywhere in-app, and I feel uncomfortable with Stats constantly sending my IP address to your server, seemingly at random. Can you please explain why this happens and what the data retention policy around any collected data is? Also, can you please provide an option to opt-out of this data collection if, like me, someone is concerned about their computer constantly pinging your server?

A screenshot of the detailed behavior is provided below:
[image]

**Details:**
 - Device: iMac (Late 2014)
 - macOS: 12.6.4
 - Application version: 2.8.14

exelban, 2023-03-30:
you need to disable ip v4 auto refresh in the network settings.

mariobrostech, 2023-04-02:
@exelban I've already done that, but the behavior persists. Your reply still doesn't address why Stats is connecting to your website to perform these checks, nor why it continues to do them when the relevant settings are disabled. I've attached a screenshot of my network settings below for completeness:
[image]
Can you please address why this is needed and what can be done to fully disable these checks? Thanks in advance!

exelban, 2023-04-02:
it's to obtain an external IP and show it in the app. You could find a description in the issues for that.

mariobrostech, 2023-04-02:
@exelban Thanks for the clarification! However, you're still not saying whether or not there's a way to opt-out of this functionality and why that might be the case. If I have the "Auto-refresh public IP address" option disabled, why is Stats still trying to ping out to obtain that information? Or, am I misunderstanding?

Also, do you mind linking me to the issues that have the descriptions of this functionality detailed in them? Because, upon an initial search, I can't find them.

Thanks again!

exelban, 2023-04-02:
disable the updates also.

mariobrostech, 2023-04-02:
Thanks for checking, but I already had updates disabled since I manage all of my packages (including Stats) using Homebrew. I've disabled all of the functionality that I can realistically attribute to requiring network access, so I'm not sure why connection requests are still being made. Is this intended behavior or a bug?

mariobrostech, 2023-04-03:
That seems to have solved the issue; thanks! Still, though, I like using the network module to gauge network traffic, which works even without the permission granted for Stats to use `curl` to connect to the network. Is this possibly a bug still, or is there no way of working around this beyond simply denying network access using `curl`?

mariobrostech, 2023-04-05:
@exelban I did some digging yesterday and have determined that this is most likely occurring due to Stats checking for the public IP address to show in the Network module. As far as I can tell, there's no way to disable this check (so, when I was testing, I simply commented out the line calling the `getPublicIP()` function).

To recap, here's the behavior I'm experiencing and how you can reproduce it:
1. With automatic updates and the ICMP connectivity checks both disabled, run a system-level firewall such as LuLu (which is the one I used, pictured above) or Little Snitch.
2. Observe that, by default, Stats tries to run the `curl` command with arguments pointing to your API endpoint, [https://api.serhiy.io/v1/stats/ip](https://api.serhiy.io/v1/stats/ip).
3. With the `getPublicIP()` function disabled, this no longer occurs.

For your reference, here are links to lines in the source code that I observed to be related to this behavior:
1. [https://github.com/exelban/stats/blob/master/Modules/Net/readers.swift#L367](https://github.com/exelban/stats/blob/master/Modules/Net/readers.swift#L367) houses the `getPublicIP()` function that makes calls out to your API ([https://api.serhiy.io/v1/stats/ip](https://api.serhiy.io/v1/stats/ip)) when invoked.
2. [https://github.com/exelban/stats/blob/master/Modules/Net/readers.swift#L413](https://github.com/exelban/stats/blob/master/Modules/Net/readers.swift#L413) references the function call that I was able to comment out to stop this behavior.

To address this, I suggest adding a toggle that would allow one to disable public IP address checking when the Network module is enabled. Stats already handles this situation by displaying "Unknown" in the Network module when it isn't able to establish a connection, so a toggle disabling this check should fail gracefully without any other changes needing to be made. Does this seem like a reasonable solution to implement for those who are more privacy-conscious and don't want any outbound network connections being made by Stats?

Thanks again for your help!

exelban, 2023-04-05:
There is the only option to disable auto updates. On run the application still will check the ip.

exelban, 2023-04-05:
There is only one option to prevent fetching the IP - disabling the Network module. I'm pretty sure that in this case fetching the public IP address will not be triggered.

mariobrostech, 2023-04-05:
@exelban I am requesting to add an option to disable the public IP address section of the network module. I still want to be able to see which applications are using my network and how much they're uploading, and this functionality still works even without access to my public IP address.

This is what I did to test disabling this behavior:
```patch
diff --git a/Modules/Net/readers.swift b/Modules/Net/readers.swift
index 9c8dcf43..6bb1d2ba 100644
--- a/Modules/Net/readers.swift
+++ b/Modules/Net/readers.swift
@@ -411,7 +411,7 @@ internal class UsageReader: Reader<Network_Usage> {
         self.usage.raddr.v6 = nil
         
         DispatchQueue.global(qos: .background).async {
-            self.getPublicIP()
+            // self.getPublicIP()
         }
     }
```

Could this be implemented as a user-facing toggle for the Network module for those who desire this functionality?

dylanarmstrong, 2023-11-13:
Hello, this is still occurring in version 2.9.11.

Chamiu, 2024-06-16:
`curl ipecho.net/plain; echo`

I don't know at all the need to access the web site and check the IP address.

It would be good if declare "no log" in Readme.md or privacy policy.