exelban/stats #1386: Bug in code: Duplicate condition in reader.swift Line 100

LostInCompilation, 2023-03-31:
**Describe the bug**
In Kit/read.swift Line 102 and Line 106 is a duplicate condition, which leads to compile errors. I don't think this is intended, please have a look into it.

```swift
            if self.value == nil && value != nil { <---- HERE
                self.ready = true
                self.readyCallback()
                debug("Reader is ready", log: self.log)
            } else if self.value == nil && value != nil { <---- AND HERE, SAME CONDITION CHECK
                if self.nilCallbackCounter > 5 {
                    error("Callback receive nil value more than 5 times. Please check this reader!", log: self.log)
                    self.stop()
                    return
                } else {
                    debug("Restarting initial read", log: self.log)
                    self.nilCallbackCounter += 1
                    self.read()
                    return
                }
            }
```

exelban, 2023-03-31:
compile error?

LostInCompilation, 2023-03-31:
Yes XCode complains about this:

[image]

exelban, 2023-04-01:
very strange, I needed to update a swiftlint to see these warnings. Already fixed.

closed 2023-04-02 by exelban