exelban/stats #1388: Menu bar indication problem

71m383nd3r, 2023-03-31:
Stats problem - battery 80% but the top menu bar indication is 89%

**Details:**
 - Device: MacBook Pro 13" (M1)
 - macOS: macOS Monterey (12.6.3)
 - Application version: 2.8.14

[image]

exelban, 2023-03-31:
popup window must be opened when updating the data. In this case 80% it's an old read.

71m383nd3r, 2023-04-01:
> popup window must be opened when updating the data. In this case 80% it's an old read.

No, it's the right value.
I'm using AlDente Charge Limiter
https://github.com/davidwernhart/AlDente-Charge-Limiter
A tool to limit maximum charging percentage, and it's set on 80%.

exelban, 2023-04-01:
what if you restart the app?

71m383nd3r, 2023-04-01:
> what if you restart the app?

it came back in sync 70% menu bar and modul.
is it a glitch?

exelban, 2023-04-01:
in 99% the value in the menu bar is right. Because it's always updates when macOS send notification about battery level change.

closed 2023-04-02 by 71m383nd3r