exelban/stats #1389: Recent update - spacing in menu no longer compact

drawingthesun, 2023-04-01:
Hi

Since upgrading to version 2.8.14 the spacing for the widgets in the menu bar are now much wider than before.

Previously they were quite compact and had very little spacing between them.

[image]

The issue could be with another app managing the icons if not an issue with stats, however all of my other icons are as they were before.

exelban, 2023-04-02:
there was no any changes in the app in the latest version that could make behavior.

closed 2023-04-02 by exelban

drawingthesun, 2023-04-03:
Sorry about that. The issue was with another app and a reboot fixed the issue.