exelban/stats #1390: [Feature Request] Add option to show CPU load in the widget

jacix, 2023-04-02:
Hello @exelban.  Thank you in advance for one of my favorite new apps.   I've been an iStat user for years but when I got a new Mac a few months ago I was out of licenses so I installed Stats on it and it continues to impress me more with every new release.

Would it be possible to add an option to the CPU widget to report the load rather than the % usage, akin to  "_w_", "_uptime_", "_top_", etc?   Even better would be the ability to select which reading (1min, 5min or 15min) to show. 

I looked at the code to see if I could do it myself (and raise a PR to share, of course) but my Swift Fu is weak.

Thanks again!

exelban, 2023-04-02:
nope, there will be not option to show that.

closed 2023-04-02 by exelban

ajeetdsouza, 2024-10-05:
@exelban I would also like this feature. Is there a reason you don't want to implement it?

scottrbaxter, 2024-11-11:
Load Average is an incredibly useful metric, which is also quite different compared to CPU Utilization. @exelban would you mind explaining what the disinterest is with adding load average in a future release?

exelban, 2024-11-11:
It's possible now since the widget that allows to show that is available. I will add that in the near future.