exelban/stats #1395: Mac m1 pro  Program terminated while running

Cwonseok, 2023-04-04:
Program terminated while running

2021 14 Mac M1 Pro (ram 32)
macOS Ventura version 13.3

The program terminates after about a few minutes while using stats.

exelban, 2023-04-04:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

Cwonseok, 2023-04-05:
[아카이브.zip](https://github.com/exelban/stats/files/11154093/default.zip)

Sorry for not providing the file in advance, I have attached it as a zip attachment.

exelban, 2023-04-05:
do you have enabled widget that have enabled monotype option?

Cwonseok, 2023-04-05:
Sorry I don't know what you mean by monotype

exelban, 2023-04-05:
Sorry, not monotype but monospaced. Please share the screenshot of your menu bar items of Stats

closed 2023-04-09 by exelban

Cwonseok, 2023-04-09:
![ Uploading 스크린샷 2023-04-10 오전 7.54.35.p [image] ng...]()

미안해 나의 mac pro 가 회사에 있어서 답장이 늦었어 여기 캡쳐본을 첨부했어