exelban/stats #1398: [Feature Request] display the uptime in the CPU menu

gianpaj, 2023-04-06:
Similar to iStat menu

[image]

Sometimes it's useful to reboot a Mac computer, and I generally check the uptime to know if it's been over a week :)

Thanks a lot for this great app! 🙏

exelban, 2023-04-06:
there is uptime in the dashboard

gianpaj, 2023-04-06:
found it . thanks 
[image]

[image]

mx1up, 2024-04-12:
as @gianpaj points out, it's not very discoverable. 

You have to open one of the widget you already have, click on the cog to go to settings, click on dashboard (because it's not selected by default), then scroll down (because the window is not resizable)

That is 4 actions vs 1 if it could be shown in one of the menu popups (like in iStat).

closed 2024-04-28 by exelban