exelban/stats #1400: Stats crash after boot

valynor, 2023-04-07:
Stats crashed after directly after booting up. 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [1148]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.8.15 (468)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-07 21:37:10.0176 +0200
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Anonymous UUID:        4016461C-24F3-2405-3768-F72B593C517E

Time Awake Since Boot: 79 seconds

System Integrity Protection: enabled

Crashed Thread:        13  Dispatch queue: eu.exelban.Stats

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000755a7dd1bdb0
Exception Codes:       0x0000000000000001, 0x0000755a7dd1bdb0

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1148]

VM Region Info: 0x755a7dd1bdb0 is not in any region.  Bytes after previous region: 23477865266609  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)
--->  
      UNUSED SPACE AT END

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a99c9c78 __bsdthread_ctl + 8
1   libsystem_pthread.dylib       	       0x1a9a068d4 _pthread_qos_override_end_direct + 32
2   libdispatch.dylib             	       0x1a9866cdc _dispatch_main_queue_drain + 1156
3   libdispatch.dylib             	       0x1a9866848 _dispatch_main_queue_callback_4CF + 44
4   CoreFoundation                	       0x1a9b27c88 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
5   CoreFoundation                	       0x1a9ae54a8 __CFRunLoopRun + 1992
6   CoreFoundation                	       0x1a9ae458c CFRunLoopRunSpecific + 612
7   HIToolbox                     	       0x1b3319df4 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1b3319c30 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x1b3319988 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x1acd03f58 _DPSNextEvent + 636
11  AppKit                        	       0x1acd030f4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 716
12  AppKit                        	       0x1accf7558 -[NSApplication run] + 464
13  Stats                         	       0x100a89dbc 0x100a84000 + 23996
14  dyld                          	       0x1a96aff28 start + 2236

Thread 1::  Dispatch queue: com.apple.root.background-qos
0   libsystem_kernel.dylib        	       0x1a99c9cd0 __ulock_wait + 8
1   libdispatch.dylib             	       0x1a9858df0 _dlock_wait + 56
2   libdispatch.dylib             	       0x1a9858d20 _dispatch_once_wait + 112
3   IconServices                  	       0x1b6cda188 +[ISIconManager sharedInstance] + 80
4   IconServices                  	       0x1b6cda040 -[ISIconFactory initWithBinding:] + 60
5   AppKit                        	       0x1ace7b864 -[NSImage _initWithIconRef:] + 132
6   AppKit                        	       0x1acf1c5d4 -[NSWorkspace _iconForURL:] + 120
7   AppKit                        	       0x1acf1c514 -[NSWorkspace iconForFile:] + 72
8   AppKit                        	       0x1ad5fbf3c -[NSRunningApplication icon] + 116
9   RAM                           	       0x100c0a7a8 0x100c00000 + 42920
10  RAM                           	       0x100c08e68 0x100c00000 + 36456
11  Foundation                    	       0x1aadd18c4 partial apply for closure #1 in StringProtocol.enumerateLines(invoking:) + 44
12  Foundation                    	       0x1aadcddf4 thunk for @escaping @callee_guaranteed (@guaranteed String, @unowned UnsafeMutablePointer<ObjCBool>) -> () + 76
13  Foundation                    	       0x1aaab1f30 -[NSString enumerateSubstringsInRange:options:usingBlock:] + 1596
[... 610 lines cut ...]
    {
      "rolloutId" : "62c751b6bcb0435c2153fe92",
      "factorPackIds" : {
        "SIRI_UNDERSTANDING_ASR_ASSISTANT" : "640bac73c473411a19effff2",
        "SIRI_UNDERSTANDING_MORPHUN" : "63f00b1c06644d312600c7bd"
      },
      "deploymentId" : 240000157
    },
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000255
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000012
    }
  ]
}
}

exelban, 2023-04-07:
please upload the file

valynor, 2023-04-07:
I had the same kind of crash a couple weeks ago: Stats crashed immediately after launch within 1-2 seconds. 
Never crashes in normal operation. No issues with other software crashing either. 

I've been using Stats for a year now and it never behaved like that before.

exelban, 2023-04-07:
please disable the disk module and monitor if it prevents the crash.

valynor, 2023-04-07:
I can do that but with the crash being very infrequent, it will take a long time (1-2 months) to get meaningful data from this.

exelban, 2023-04-07:
I have no idea what is the reason for the crash. The crash report does not give information on what was the reason. Only the Disk module is noticed. But I think it's only because in crash moment this module was active.

omarsonmarz, 2023-04-08:
Also have this problem, has crashed 3 times in the past hour and a half. 
[image]
Here are the console logs for each crash, crashlog3 being the first crash :
[crashlog3.txt](https://github.com/exelban/stats/files/11183733/crashlog3.txt)
[crashlog2.txt](https://github.com/exelban/stats/files/11183734/crashlog2.txt)
[crashlog1.txt](https://github.com/exelban/stats/files/11183735/crashlog1.txt)

omarsonmarz, 2023-04-08:
I have noticed that it does crash sometime after a wake from sleep, however according to the logs this "crash since wake” time increases with each crash e.g. 1st crash was 174 seconds (about 3 minutes) since wake while the latest crash was 11744 seconds (about 3 hours) since wake.

exelban, 2023-04-08:
@omarsonmarz do you have an enabled the monospaced option in any widget?

omarsonmarz, 2023-04-08:
> @omarsonmarz do you have an enabled the monospaced option in any widget?

Yes, for the sensors widget which includes the fan speed, Avg CPU and Avg GPU temps.

exelban, 2023-04-08:
yeap. it's a reason. Please disable this feature.

exelban, 2023-04-09:
duplicate of #1395

closed 2023-04-09 by exelban

valynor, 2023-06-19:
@exelban Just a quick feedback: no more crashes since I originally reported this issue.