exelban/stats #1402: Stats randomly crashes in the background.

coldenate, 2023-04-09:
Hi! I have noticed stats will just disappear from my menubar and I will have to restart it. What is the process to take to give you the logs?

exelban, 2023-04-09:
Do you have the latest version installed?

coldenate, 2023-04-09:
I do. I am currently waiting to see if the behavior repeats. It may be a false positive since I cannot tell when it is updated. I had this issue yesterday mostly.

exelban, 2023-04-09:
Please check what exactly version of Stats you have

coldenate, 2023-04-09:
Version 2.8.16

coldenate, 2023-04-09:
I believe updating fixed the issue. I will let you know if I see it happen again.

exelban, 2023-04-09:
ok, give me to know.

exelban, 2023-04-10:
looks like it's fixed

closed 2023-04-10 by exelban

bemfica, 2024-06-07:
Not sure if it's related to this, but it was the closest issue I found with what I'm experiencing. Stats sometimes just closes unexpectedly, no crash window, it just disappears from the menu bar. Tried disabling storage widgets as suggested in other issue, didn't help. Starting Stats from the terminal gives me the following message when the modules disappear:
`Object` 0x[RANDOM HEXADECIMAL] `of class _ContiguousArrayStorage deallocated with non-zero retain count 2. This object's deinit, or something called from it, may have created a strong reference to self which outlived deinit, resulting in a dangling reference.`