exelban/stats #1403: Disk Usage top processes not visible

FrostBird347, 2023-04-10:
**Describe the bug**
No processes appear in the disk usage module.
[image]

**Details:**
 - Device: MacBook Air (Retina, 13-inch, 2019)
 - macOS: 10.14.6
 - Application version: 2.8.16

exelban, 2023-04-10:
because there is nothing to show at this moment.

FrostBird347, 2023-04-11:
Are you saying that processes should appear eventually, or that you haven't added this functionality yet?
Because if it's the former then something is wrong as I haven't seen any processes appear after waiting for a while with the menu open while also compressing files/

exelban, 2023-04-11:
I can send you a debug build if you would like to run it and debug the reason why it's not showing.

FrostBird347, 2023-04-12:
I am alright with using a debug build

exelban, 2023-04-12:
There is a debug build with some logs that will be written to `Documents/log.txt`. Please run this build with opened Disk popup window and share the logs after.
[Stats.dmg.zip](https://github.com/exelban/stats/files/11215141/Stats.dmg.zip)

Loganpoor, 2023-04-13:
> 

I found this debug build also fixed the wrong SSD free size issue.
Thank you very much!

FrostBird347, 2023-04-13:
> There is a debug build with some logs that will be written to `Documents/log.txt`. Please run this build with opened Disk popup window and share the logs after. [Stats.dmg.zip](https://github.com/exelban/stats/files/11215141/Stats.dmg.zip)

Here's the log: [log.txt](https://github.com/exelban/stats/files/11221052/log.txt)

closed 2023-04-13 by exelban

johnalanwoods, 2023-05-08:
Do we know the root cause?

johnalanwoods, 2023-05-09:
So this should work in Ventura only?

exelban, 2023-05-09:
I'm not sure. Without logs I cannot help.

johnalanwoods, 2023-05-09:
Happy to post logs, trying to understand what’s going on with this issue. 

Why was the issue closed when it’s still an issue for people?

I have this issue on an Intel, and 2 ARM Macs, across a number of versions of macOS. 

I see processes in CPU and RAM tabs. I don’t get it?