exelban/stats #1404: [Feature request] Show label on top row of clock module and add AM/PM toggle

GiorgioMendieta, 2023-04-10:
Instead of only showing the time in the format specified (ex. hh:mm:ss) it would be nice to also show the label on the top row in bold.

- Menu bar example (before):
12:14

- Menu bar example (after) : 
**Paris**
12:14

Another useful feature would be to show AM/PM indicators next to the time. Maybe a toggle on the widget settings tab.

- Menu bar example : 
12:14 a.m. / 12:14 AM

I would try working on it but I don't have experience developing apps for mac, if you can give me some pointers i could try!

Thanks!

exelban, 2023-04-11:
Thanks for the feedback.
You can add AM/PM with `a`.

exelban, 2023-04-12:
And label will not be added because it's not possible in this widget.

closed 2023-04-12 by exelban