exelban/stats #1410: Combined modules not working on Ventura 13.3.1 (22E261)

qzhsjz, 2023-04-14:
**Describe the bug**
When enable "Combined modules", there's no widgets shown on the menu bar.
Before enabled "Combined modules":
[image]
After enabled "Combined modules":
[image]

**Details:**
 - Device: MacBookPro18,4
 - macOS: Ventura 13.3.1 (22E261)
 - Application version: 2.8.16

exelban, 2023-04-14:
Please leave only the CPU module enabled and recheck the combined option.

qzhsjz, 2023-04-15:
Well now i see how it works, thanks.

closed 2023-04-15 by qzhsjz