exelban/stats #1412: Bluetooth Icon is a no entry sign?

alexleach, 2023-04-14:
**Describe the bug**
I've just enabled the Bluetooth module in Stats, and the icon is a no-entry sign. Further, it doesn't seem to show the battery level of my headphones. Is there anything I need to do to have the icon displaying properly? I had a browse through the issues, and have similar output to https://github.com/exelban/stats/issues/1300, where `/usr/sbin/system_profiler SPBluetoothDataType` doesn't show the battery level percentage of my device... Nor does `ioreg`, as far as I can see...

Thanks,
Alex

[image]

**Details:**
 - Device: MacBook Pro 14" M2 2023
 - macOS: Ventura 13.3.1
 - Application version: 2.8.16

exelban, 2023-04-14:
Stats do not have a battery level. Please see another issue where that was described.

closed 2023-04-14 by exelban

alexleach, 2023-04-14:
Thanks for the quick reply. So, the bluetooth icon is supposed to be a no entry sign?

exelban, 2023-04-14:
yes, if there is no data to show.