exelban/stats #1416: Allow hiding "Install fan helper" placeholders from Sensors popup

wzxu, 2023-04-16:
**Describe the bug**
I understand the placeholders are there to remind users it's possible to further enable fan control as an optional feature. However, if they already use other apps like Macs Fan Control, or simply don't need that feature, they're stuck with these placeholders every time they open the popup.
[image]

I propose a way to allow the user to dismiss these placeholders. If they want to install the helpers again later, it may be possible to do so from within the Sensors tab in settings.

**Details:**
 - Device: MacBook Pro 14" (M1 Pro)
 - macOS: macOS Ventura (13.3.1)
 - Application version: 2.8.17

exelban, 2023-04-17:
Yeap, I think this button could be moved.

valynor, 2023-04-23:
Maybe use this opportunity to display both % and/or RPM for the fans in the popup?

closed 2023-05-07 by exelban

NostromoUS, 2025-01-19:
I still see the placeholders.