exelban/stats #1417: Feature Request: Clock improvements

dabeeeenster, 2023-04-17:
A few things that I would love!

- To be able to reorder the clocks in the settings page to control their order in the popup
- To be able to define cities/regions rather than UTC+N - Presuming UTC+N is not timezone aware to the region? 

Also it's not clear to me what this does? 

[image]

Thanks again for the tool!

exelban, 2023-04-17:
Hi. Thanks for the feedback.
Yeap, I think it could be done. Not sure about the city/region rather than UTC, but will check.

I don't understand. You mean: what does local time mean?

dabeeeenster, 2023-04-17:
> I don't understand. You mean: what does local time mean?

I've no idea what this UI element does?

exelban, 2023-04-17:
your local time

closed 2023-04-30 by exelban