exelban/stats #1418: Connectivity history doesn't display anything

LostInCompilation, 2023-04-17:
This is more of a question, however the discussion function of GitHub is not available for this repo.

The new connectivity history seems to not show anything. My Mac was turned on the last two days and connected to a network via WiFi. I'm not sure how this new functionality works. Is it normal that they are just grey boxes?

[image]

exelban, 2023-04-17:
Do you have an enabled connectivity check?

exelban, 2023-04-17:
Connectivity host is empty, so Stats is not check the connectivity.

LostInCompilation, 2023-04-18:
Oh OK, I thought this would just read the systems info if for example WiFi is connected. Getting the status with ICMP is way more flexible. I will try it.

closed 2023-04-18 by LostInCompilation