exelban/stats #1420: Monitor multiple hosts with ICMP

LostInCompilation, 2023-04-18:
Now that I understood the connectivity history feature, I found it very useful for monitoring my router.

A very useful thing would be if multiple IPs could be monitored with ICMP. Maybe a list showing all IPs and their respective connectivity history. By reducing the number of boxes displaying the history, or making a setting available to display the history in a different way, or just the current/last 30 seconds status visible, stats could be a powerful network monitor tool.
I think this could be extremely useful even for standard home user with a local NAS or Raspberry Pi with Docker and multiple servers (with different IPs).

Maybe you could consider something like that. There is much room for cool and powerful monitoring capabilities.

exelban, 2023-04-18:
I do not understand the proposal at all. Your Mac have multiple IPs and you want to monitor all of them or what?

LostInCompilation, 2023-04-18:
I have multiple servers in my local network, each with their own IP. Monitoring these via ICMP is the the point. That would require the ability to input multiple connectivity hosts.

LostInCompilation, 2023-04-18:
But you can use it for monitoring local servers. Any IP with ICMP support works. For example I just entered the IP of my router in the settings.

exelban, 2023-04-18:
yes, you can use this way to monitor almost any resource that uses IP. But this application is not for that. 
It's the best way that I know to determine if there is a connection with the internet (via pinging some external IP).

closed 2023-04-18 by exelban

LostInCompilation, 2023-04-18:
Yes thats why I wrote this proposal. Stats could become a very useful tool for monitoring local servers/your home network infrastructure by allowing multiple ICMP hosts and displaying multiple hosts in a different way in the pop up.

LostInCompilation, 2023-04-18:
I would appreciate it if you could leave this issue open for a bit to get some feedback from more users what they think of it. I would expect it to be useful to have monitoring built right into stats.

exelban, 2023-04-18:
It will be not added to the Stats. Because it's a macOS monitoring. 
It is an idea for a dedicated app. But there are apps that do that.