exelban/stats #1421: Allow "no space" between widgets on the menu bar when the widgets are NOT combined

0xAdriaTorralba, 2023-04-18:
**Describe the bug**
When the option "combined modules" is ticked, there is a dropdown menu that allows to set "no space" between widgets on the menu bar. This option does not appear when the "combined modules" is not ticked and by default it applies some spacing to them.

**Details:**
 - Device: MacBook Pro 16 inch M1 Max
 - macOS: 13.3.1
 - Application version: 2.8.17

0xAdriaTorralba, 2023-04-18:
Combined modules is ticked:
[image]
"No space" between modules:
[image]

Combined modules is NOT ticked:
[image]
Default (not modifiable) space between modules:
[image]

exelban, 2023-04-18:
Yes, because these option is only for the `combined mode`.

closed 2023-04-18 by exelban

0xAdriaTorralba, 2023-04-18:
Can it be possible to implement it outside `combined mode`?