exelban/stats #1422: External keyboard and trackpad/mouse freeze on wake up from sleep

hackentogler, 2023-04-18:
I have tried with only apple Bluetooth keyboard/mice, Logitech only using the unifying receiver, and a mix. The only consistency is when waking M1 MacBook Pro 16” from sleep (sometimes right away but always occurs when sleeping overnight) the external keyboard and mice/trackpads are unresponsive for 10 to 20 seconds. The keyboard and trackpad on the MacBook Pro ARE responsive during this time. It does not matter if the MacBook is open or in clamshell, the external device delay still occurs. Locking the screen and coming back to it causes no delay at all. Only waking from sleep by plugging it 5k Lg ultra fine and 4k Dell P2415q. Both are connected to their own USB-C port. There is a Time Machine Drive and Logitech bolt receiver on back of 5k. However, this problem occurs even using only Apple Bluetooth keyboard/trackpad/mouse as well. I’m running the memory, network, gpu and cpu monitors in the menu bar.

exelban, 2023-04-18:
emm, so?
I'm not sure I understand how this history is related to this app.

hackentogler, 2023-04-18:
It only happens when the app is running. If I quit the app, none of the
delays waking up occur.

exelban, 2023-04-18:
strange. Since you don't even have enabled Bluetooth module. Could you try to disable one by one modules and determine what causes the problem?

exelban, 2023-04-20:
I check that, and have no idea what cause freeze. Especially if the GPU module causes that. They're nothing special inside, only read from IO registry. Could not help with that.

closed 2023-04-20 by exelban