exelban/stats #1433: can't control fan speed

zerocoolwick, 2023-04-24:
can't control fan speed,
but the other sensor works well.

[image]

exelban, 2023-04-25:
Most probably restarting the app will help. Otherwise, I cannot help with that.

exelban, 2023-04-25:
Hmm. Please check if you have running `eu.exelban.Stats.SMC.Helper ` when running Stats. It might help to understand why it does not work.

zerocoolwick, 2023-04-26:
[image]

The GUI control can't change fan rpm, using CLI works well.

then it will show the value of CLI setting speed on the top widget, but doesn't match the GUI speed me adjusted.

zerocoolwick, 2023-04-26:
[image]

CLI modify fan cotrol to 1 ( manual) then uninstall fan helper and re-install it again,
The GUI adjust page works, even changing CLI fan control to 0.

Thanks for the respones