exelban/stats #1434: App icon proposal

Zabriskije, 2023-04-24:
Hey Serhiy,

As a thank you for making Stats, I've reworked the current icon to give it a more modern look and corrected the frame to be on par with the one found in Activity Monitor, Terminal, and Console. Since I'm a big fan of minimalism, I also made one without the frame. 
Left to right: current one, reworked, minimal.

[image] [image] [image]

Both are 1024x1024 and ready to be used.

LostInCompilation, 2023-04-24:
The right one would be my favourite.

Zabriskije, 2023-04-24:
> The right one would be my favourite.

I've edited my message and added a ZIP containing both PNG and ICNS files, so you can use it right away if you want to ;)

LostInCompilation, 2023-04-24:
Thanks but I'm not the owner 😅 @exelban needs to decide.

But I think the current trend for modern icons fits well with the right one.

Zabriskije, 2023-04-24:
> Thanks but I'm not the owner 😅 @exelban needs to decide.
> But I think the current trend for modern icons fits well with the right one.

Oh, I know. It was just to make it available already packed as ICNS, so everyone can use it no matter what  :)

exelban, 2023-04-24:
Hi. Thanks for your work. I like the right one most)
But it feels a little bit stretched in width. Looking at him, I feel like it's not a square but a rectangle. Could it be changed in some way?

LostInCompilation, 2023-04-24:
@exelban The right icon is a perfect square. When cropped to content only, the dimensions are 824x824. Looks like it's just an illusion.

Zabriskije, 2023-04-24:
@exelban Now that you've pointed it out, I see that it appears a bit stretched here. It's not that obvious when looking at it with other icons in Launchpad/Dock. Does resizing the lines inside help?

[image] [image]

exelban, 2023-04-24:
yeap, a little bit. Could you please share the pack of assets for the test?

Zabriskije, 2023-04-24:
[Here's the .iconset](https://github.com/exelban/stats/files/11313622/Stats-Iconset.zip)

exelban, 2023-04-24:
Thanks. I will try to live with that for some time and will give feedback. Thanks again)

PS: if anyone wants to see how it looks there is a build with a new icon: 
[Stats.dmg.zip](https://github.com/exelban/stats/files/11313720/Stats.dmg.zip)

LostInCompilation, 2023-04-24:
+1 for the new icon in next release.

BitesPotatoBacks, 2023-04-25:
Those look awesome. +1

liuzhihang, 2023-04-27:
Can I adjust the background color of the panel?

Zabriskije, 2023-04-27:
@liuzhihang Are you asking for the source file? If that's the case, it's a Pixelmator `.pxd`, I don't know if Photoshop or other software can open it correctly, but [here it is](https://github.com/exelban/stats/files/11341989/Stats.zip).

wzxu, 2023-04-29:
Great work. Have been applying iStats' icon over the current one because that corner radius mismatch is killing me. One less icon to change is always welcome 💯

exelban, 2023-04-30:
the new icon is available in the new release. @Zabriskije thanks.

closed 2023-04-30 by exelban

Zabriskije, 2023-04-30:
You're welcome!
May I get a little mention somewhere? 👉👈

exelban, 2023-04-30:
yeap