exelban/stats #1437: 2 bugs in disk module

valynor, 2023-04-25:
Mac Studio, Ventura 13.3.1
iMac 2017, 13.3.1
Stats 2.8.18

1. I noticed some discrepancy in the behaviour of the Disk popup -> Top processes: 
On the iMac the Top processes update every second, on the Mac Studio they update only every 10 seconds
Both modules are set up exactly the same: 5 Top processes, 10 sec update interval. 
Changing the update interval on the Mac Studio seems to have no effect (stays at 10 seconds). 

2. Mac Studio: When I first used 2.8.18 yesterday the Temperature for "Macintosh HD" in the Disk popup was the same (but rounded up) as what was displayed in the Sensors popup -> NAND. 
Today the temperature is stuck at "22°C" and does not change at all. Sensors -> NAND shows 26.0°C and moves up and down a bit as expected.

LostInCompilation, 2023-04-25:
Just wanted to report the first issue as well.
The Update Interval setting has no effect on the Drive widget. The values in the menu bar update every second.

exelban, 2023-04-25:
1. There is no option in the app to change the top processes update interval.
2. There is no relation between the temperature in the Disk module and any of sensors in the Sensors module.

closed 2023-04-25 by exelban

valynor, 2023-04-25:
1. there's a clear difference between 2 machines with the same settings
2. 2. the temp value didn't change for a whole day

there are clearly some bugs here, even if you apparently chose to ignore them

exelban, 2023-04-25:
1. You think that there is some difference. But there cannot be any difference. The reader runs every 1s. The only difference is in `ps` output. If there is no data for a few seconds, or minutes you will see no changes in the popup. It's a reason why I refuse for such a long time to implement that. Because I have no impact on how that works. And after I must explain each time why "it does not work as someone expect".
2. It could stack only if the NVMe SMART plugin does not return the data.

valynor, 2023-04-30:
With the latest update 2.8.19 the disk popup now updates the top processes every second on my Mac Studio.

exelban, 2023-04-30:
Hi. Yeap, you were right. There was a bug. Unfortunately, I was testing that on the build that has a fix. And didn't notice that(