exelban/stats #1440: Update interval setting has no effect on Disk Widget

LostInCompilation, 2023-04-25:
**Describe the bug**
Since you closed the previous issue but somehow missed my comment, I open a separate issue here.

Changing the `Update interval` setting for the Disk widget does not change the update speed of the widget in the menu bar. It's always updating once a second. (Note: I do NOT mean the top processes, just the widget)

**Details:**
 - Device: MB Pro 2021 16"
 - macOS: 13.3.1
 - Application version: 2.8.18

exelban, 2023-04-25:
I to recheck that. Everything works as should. Update interval changes the read interval of the disk capacity reader.

exelban, 2023-04-25:
Please do not duplicate the issues that I already answer. Especially if it was a few minutes ago, and you have read the answer.

closed 2023-04-25 by exelban

LostInCompilation, 2023-04-25:
This was your answer:
>There is no option in the app to change the top processes update interval.

Nothing about update interval of R/W speeds shown in menu bar.
But anyway, can an update interval for R/W speeds be added?

I mean those:
[image]

exelban, 2023-04-25:
Nope, there is no option to change this interval. See an issue related to the network IO. 
TL.DR: the reason why you see the value: MB/s it's because it's an IO stat per second. And not per 10s. Otherwise, I need to show MB/10s.

exelban, 2023-04-25:
Which has no sense to me and provides a lot of confusion.

LostInCompilation, 2023-04-25:
I meant updating, and therefore reading the R/W speeds from system only every N seconds. The time where no update happens is just lost information. Like it is with CPU Interval, RAM interval, Sensor interval.

For example:
Update interval for Disk set to 2 seconds.
Second 0: Read from system as display.
Second 1: Do not read from system, display previous value.
Second 2: Read from system as display new values
...

exelban, 2023-04-25:
It will not work. Because I will have no last read value.
0s - I read 100mb IO
10s - I read 230mb IO
20s - 540mb IO
...etc
How to calculate what I need to show?

Unfortunately, there is no way option to have such a lot of information and not pay for that by power consumption. As I wrote in the Readme: if you want to reduce energy impact from the Stats you need to disable some Stats modules. And there is no option to reduce that. I'm trying to keep CPU usage as low as possible, but it's not always possible.

LostInCompilation, 2023-04-25:
Okay I thought you receive the data per seconds directly from kernel, not only the amount of data.

exelban, 2023-04-25:
Nope, there is only one way I know.

exelban, 2023-04-25:
Also, it's quite cheap way. I'm pretty sure that the CPU impact of this reader is around 0.1%. A lot (like a LOT) less that capacity/temperature reader.

mailinglists35, 2024-01-26:
> How to calculate what I need to show?

by looking at how others do it?
how does windows task manager updates every 5 seconds?
how does mac activity monitor calculates it when update frequency is 5 seconds?
how do top, atop and nmon command line utilities calculate when update frequency is 5 seconds?

mailinglists35, 2024-01-26:
glances also show disk i/o metrics https://nicolargo.github.io/glances/