exelban/stats #1441: Fan RPM in manual mode only show up after changing the fan speed

LostInCompilation, 2023-04-25:
**Describe the bug**
When switching the fans from Automatic to Manual, the current fan RPM doesn't show up until I change the speed of the fans.
I would expect the RPM value to show up instantly when switching to Manual mode.

Here's a short video: https://youtube.com/shorts/uOqsqFphldw

**Details:**
 - Device: MBPro 2021
 - macOS: 13.3.1
 - Application version: 2.8.18

exelban, 2023-04-25:
Yes, it works like that. It's indicated the target fan speed.

closed 2023-04-25 by exelban

LostInCompilation, 2023-04-26:
IMO that's a bit unintuitive. User would normally expect the current Fan RPM value, which should always be displayed. Why is the target fan speed (which should always be the same as the current fan speed) by of interest?