exelban/stats #1448: GitHub Desktop Network Activity Not Showing

GetzPro, 2023-05-04:
**Describe the bug**
Github desktop showing download activity (and I hear the HDD working), yet your tool, nor Activity Monitor, shows this activity in network monitoring. 

This is when I clone from Hugging Face. Some times I see very active network when downloading, but when it gets into large files, the network activity drops to 0 but Github shows activity. 

Might this be something with LFS files? 

Thank you!

**Details:**
 - Device: 2019 MBP
 - macOS: 13.3.1 (a) 
 - Application version: 2.8.19

GetzPro, 2023-05-05:
I see your disk tool measuring more writes than the network measuring downloads.

exelban, 2023-05-06:
network processes are from `nettop`. It does not depend on the stats if some processes are missing. Because it only visualizes the data.

closed 2023-05-06 by exelban