exelban/stats #1451: Disk module update interval seems to be 1 second

nsa-twell, 2023-05-05:
(Regardless of my settings.)

VivaLosDoyers, 2023-05-05:
Just a guess, but I suspect the menu icon is respecting your time setting, but when you click to open the details, the details pane is refreshing every second. The preferences does not contain a separate setting for this like it does for the top processes in CPU and RAM modules.

exelban, 2023-05-06:
The disk module respects all intervals.

closed 2023-05-06 by exelban

nsa-twell, 2023-05-09:
> Just a guess, but I suspect the menu icon is respecting your time setting, but when you click to open the details, the details pane is refreshing every second. The preferences does not contain a separate setting for this like it does for the top processes in CPU and RAM modules.

I was talking about the details window, yes, it updates every second. For the CPU & RAM ones the update interval parameter affects both menu bar icon and details.

If the disk module intentionally behaves differently, then no issue as such.