exelban/stats #1453: Feature Request: RAM Pressure

cmd2001, 2023-05-06:
On recent Macs based on Apple Silicon with limited RAM, which use swap much more aggressively, memory pressure is a more significant parameter to measure the overall load of system memory than legacy memory usage.

On macOS, we can get memory pressure is from a kernel variable called vm.memory_pressure, an 0-100 value, simply using the following command:

```sh
sysctl -a vm | grep vm.memory_pressure
```

In Swift, where the case is more sophisticated,  there is no direct method to get vm.memory_pressure, but we can use Objective-C code, which utilizes the `sysctlbyname` function to get the value of vm.memory_pressure.

```objective-c
import Foundation
import Darwin

func getMemoryPressure() -> Int32? {
    var value: Int32 = 0
    var size = MemoryLayout<Int32>.size
    guard sysctlbyname("vm.memory_pressure", &value, &size, nil, 0) == 0 else {
        return nil
    }
    
    return value
}
```

We can test previous code piece by creating a single file `memory_pressure.m` with the following code:

```objective-c
#import <Foundation/Foundation.h>
#import <sys/sysctl.h>

int main(int argc, const char * argv[]) {
    @autoreleasepool {
        int value = 0;
        size_t size = sizeof(value);
        
        if (sysctlbyname("vm.memory_pressure", &value, &size, NULL, 0) == 0) {
            NSLog(@"vm.memory_pressure = %d", value);
        } else {
            NSLog(@"Error getting vm.memory_pressure");
        }
    }
    return 0;
}
```

Compile and run it using:

```sh
gcc -o memory_pressure memory_pressure.m -framework Foundation
./memory_pressure
```

The output could be like:
```
2023-05-06 14:28:01.536 memory_pressure[19693:384755] vm.memory_pressure = 0
```

However, I am not familiar with this project.

So maybe I just need some help to integrate memory pressure value into stats project.

cmd2001, 2023-05-06:
#1383 There is no point to manually calculate memory pressure, we can just retrieve it from macOS kernel variable.

exelban, 2023-05-06:
There is no way to get the actual memory pressure value. Only with calculation. 
`memory_pressure` - does not work as expected. 

Even your example: 
`2023-05-06 14:28:01.536 memory_pressure[19693:384755] vm.memory_pressure = 0`

The memory pressure level is the only one that could be obtained from sysctl.

closed 2023-05-06 by exelban