exelban/stats #1454: Stats for MacBook Reporting Inaccurate Temperatures on macOS Devices

danielkhendricks, 2023-05-06:
**Describe the bug**
The "stats for MacBook" library is reporting inaccurate temperature values for Celsius, specifically for the CPU diode virtual temperature. Additionally, other temperature parameters are also affected, and it appears that Fahrenheit values are being reported without proper conversion to Celsius. For example, the CPU diode virtual sensor is reporting a temperature of 55C, which is much higher than the actual temperature.

**Details:**
- Device: MacBook Air 2021
- macOS: 11.5.2
- Stats for MacBook version: 2.4

**Steps to Reproduce:**
1. Install the "stats for MacBook" library.
2. Use the library to retrieve temperature data in Celsius for various parameters, including the CPU diode virtual sensor.
3. Compare the reported Celsius temperature to the actual temperature of the device.
4. Observe that the reported Celsius temperature appears to be significantly higher than the actual Celsius temperature. Additionally, Fahrenheit values are also being reported without proper conversion to Celsius.

**Expected behavior:**
The "stats for MacBook" library should accurately report the temperature values in Celsius for all parameters, and should properly convert Fahrenheit values to Celsius.

**Actual behavior:**
The library is reporting Celsius temperatures that are significantly higher than the actual Celsius temperature of the device, and is also reporting Fahrenheit values without proper conversion to Celsius.

**Additional information:**
- The issue has been observed on multiple devices running macOS.
- Other temperature monitoring tools or libraries do not report similar discrepancies in temperature values.
- The issue persists even after reinstalling the "stats for MacBook" library or updating to the latest version of macOS.
- The discrepancy is not affected by changes in usage or other environmental factors.

BitesPotatoBacks, 2023-05-06:
@danielkhendricks  A few questions...
1. What other tools are you comparing temperatures with, and how are you determining that they don't have this discrepancy?

2. You said you are comparing the CPU diode virtual temperature and the actual device temperature. By that I'm assuming you mean the laptop's physical shell? (unless you're putting your finger directly on the heatsink lol) To that comparison I would say, CPU temps are not directly reflective of device temperature, they merely influence it; for example a CPU can run at 99c while the bottom of your laptop is only be 45c. So I'm a bit confused there.

3. What _exactly_ do you mean by "Fahrenheit values are also being reported without proper conversion to Celsius"? Can you give a concrete example?

exelban, 2023-05-06:
@BitesPotatoBacks thanks. But there is no sense in discussing the version almost 2 years from now.

closed 2023-05-06 by exelban